ethz-adrl/control-toolbox #203: Cloning and extracting zip fails

MartinKoch123, 2021-08-07:
Hi, 

when cloning I get this error:

[image]

When downloading and extracting the zip file I get this error: 

[image]

Is this some kind of Linux/Windows issue? I'm using Windows.

Thanks and best regards

markusgft, 2021-08-07:
Hi, this is likely because of windows file system incompatibility with the CT
We have never tested this on windows and only support Ubuntu 18.04, and 20.04
Best

MartinKoch123, 2021-08-07:
Okay, thank you for the quick reply.

zhangbaozhe, 2022-11-07:
You may try to modify the case insensitivity on Windows. Check this [link](https://learn.microsoft.com/en-us/windows/wsl/case-sensitivity). @MartinKoch123

MartinKoch123, 2022-11-07:
Thanks for the suggestion but the issue is no longer relevant for me. Feel free to close.

connorjak, 2022-11-09:
Seems this in particular is a different issue with filenames: https://web.synametrics.com/syncrify-xtra-spaces.htm