ethz-adrl/control-toolbox #204: matplotlib import error

Zcyyy, 2021-08-08:
Error occurs when I run the “rosrun ct_core plotTest” command.
```
~/ct > rosrun ct_core plotTest                                   08:12:03
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error loading module matplotlib.pyplot!
[1]    27682 abort (core dumped)  rosrun ct_core plotTest
```
I have tried some methods，but There is no effect.

markusgft, 2021-08-08:
Likely some version mismatch!

Zcyyy, 2021-08-08:
@markusgft Thanks a lot. I run it successful when i change matplotlib version.

closed 2021-08-08 by Zcyyy

NisharnSP, 2021-11-29:
> @markusgft Thanks a lot. I run it successful when i change
what matplotlib version did you use?

Wow-zheng, 2022-07-22:
> > @markusgft Thanks a lot. I run it successful when i change
> 
> what matplotlib version did you use?

Did you solve the problem, thanks for your reply

Wow-zheng, 2022-07-22:
After testing, the solution is as follows:
Ubuntu18.04
Python3.69
matplotlib3.0.2