ethz-adrl/control-toolbox #210: How to add MPC as a member in a class?

C1009919253, 2022-03-07:
For example, I want to add MPC's object mpc_t in class A, but I can't initialize mpc_t in A's constructor function.
How could I fix that problem?

condor-cp, 2022-03-08:
I would guess the problem here is the lack of a default constructor of the MPC class. 

What you can do to fix that is for example have a shared_ptr<MPC> as your member class and dynamically define it in the constructor. 
Another option can be to call the MPC constructor in the initializer list as suggested in https://stackoverflow.com/questions/12328150/class-member-without-a-default-constructor

C1009919253, 2022-03-09:
> I would guess the problem here is the lack of a default constructor of the MPC class.
> 
> What you can do to fix that is for example have a shared_ptr as your member class and dynamically define it in the constructor. Another option can be to call the MPC constructor in the initializer list as suggested in https://stackoverflow.com/questions/12328150/class-member-without-a-default-constructor

Thanks for your help, I tested the first way, and it worked!

closed 2022-03-09 by C1009919253