ethz-adrl/control-toolbox #215: Installation error with blasfeo

dd292, 2022-06-19:
Hey,

I am just trying to run this library in Ubuntu 20.04 but getting the error:
`In file included from /usr/local/include/ct/optcon/nloc/NLOCBackendBase.hpp:16,
                 from /usr/local/include/ct/optcon/optcon.h:25,
                 from /home/airlab/Documents/Projects/Cppilqr/src/control-toolbox/ct_models/src/HyQ/codegen/HyQLinearizationCodegen.cpp:6:
/usr/local/include/ct/optcon/solver/lqp/HPIPMInterface.hpp:12:10: fatal error: blasfeo_target.h: No such file or directory
   12 | #include <blasfeo_target.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
`
I installed the HPIPM and BLASFEO using ./src/control-toolbox/ct/install_hpipm.sh

I was curious if this issue is because of my Ubuntu version or just incorrect installation.
Would appreciate the help.

condor-cp, 2022-06-28:
I had the same issue that you have but somehow the repo owner could not reproduce. Try to modify CMakeLists.txt files as I did with [this commit](https://github.com/condor-cp/control-toolbox/commit/d677bcb935d605290775e78973c42df1268a0406)

Regards

haasithp, 2022-06-30:
Hey, I was able to install and made these notes. Please go through it and let me know if there are any issues.
https://haasith.notion.site/Control-Toolbox-ec4e170501914120b4b666e89353d35c